fanoush/ds-d6 #1: Help regarding notifications on lenovo hw01/DS-D6

techieaarya, 2019-02-28:
I have A Lenovo Hw 01 And the companion app (Lenovo healthy and the new app Lenovo life) does not have an option for notification of Instagram. I found that only you can help as nobody else is interested in this band. the MPOW app when gets connected to a rebranded band sends` AT+MOTOR:1` and the band starts vibrating, every time the Bluetooth gets disconnected and reconnected the band starts vibration. I tried the Bluetooth serial terminal app as suggested by you and I found that many commands are different between our bands, after going through the Bluetooth snoop log I found a way to show custom text on the band but I want to develop an app which can help me with the notifications. I do not have any knowledge of programming and I am taking this as my part-time project. so far I have identified the syntax and certain limitations of the band. I would like to help you if possible and learn something new. further, I would like to personally talk to you [@techie_aarya](https://www.instagram.com/techie_aarya) is my Instagram id please DM me. 

**I would also like to change the we chat logo from the band to instagram logo through a OTA update with your help.** 

[MY EMAIL](mailto:techiejawab@gmail.com)

fanoush, 2019-02-28:
Hi,
according to https://fccid.io/2AEMN-HW01/Letter/Cover-Letters-3407427 this is same device as https://fccid.io/2AEMN-B521 however the internal photos there are of poor quality so I don't see cpu markings nor SWD test points and also it is not clear if usb connector also has serial port usable for DFU update like DS-D6. Date of submission to FFCID is 2016-10-23  - 7 months before DS-D6 so it is older and may even use older Nordic SDK (or even older CPU), But this is just guesswork, it may still be same as DS-D6. From MPOW app DSEBandHW.json it looks like HW01 is Nordic and commands are mostly same but there are more of them, DS-D6 has some of them removed.

For developing android app it may make sense to start with something opensource from f-droid  and extend it e.g. https://gadgetbridge.org/

For modifying firmware you either need to open it and attarch SWD debugger or you need original firmware package zip and then you need to be extra careful. Once you flash something that does not work you need to open it anyway.

I do not have instagram but we can continue here. for chat there is https://gitter.im/nRF51822-Arduino-Mbed-smart-watch/Lobby

techieaarya, 2019-03-03:
The CPU has **N52832** **QFAAB0** **1712F0** written on it, and okay I think its a risk to perform an update as I don't want to brick my band and I am not familiar with **SWD** so I think, I should not try it. but i also have a question- why does my computer recognise the band as an unknown USB device, is there any driver software that can help? [these are the photos I took which might help you(Google Drive Link) ](https://drive.google.com/open?id=1JvpDTruX4JLNNj0nBNlxDg-lEvfFqAZH). I am developing the app soon as I am working on its UI. f-droid will really help me, and again thanks for the help. I will be notifying you about the progress.

Regards.

fanoush, 2019-03-04:
Oh, so you already opened it, good. SWD pins are those IO and CLK in the corner near the GND. I guess the cheapest and easiest way to do SWD debugging is to get ST-Link V2 clone from aliexpress or ebay like eg. [this one](https://www.aliexpress.com/item/1Set-ST-LINK-Stlink-ST-Link-V2-Mini-STM8-STM32-Simulator-Download-Programmer-Programming-With-Cover/32866866559.html) and use it with OpenOCD - it is easiest in linux (e.g. on raspberry pi).
BTW, if you get two of those ST-Links you can [reprogram one with the other](http://blog.linuxbits.io/2016/02/15/cheap-chinese-st-link-v-2-programmer-converted-to-black-magic-probe-debugger/)  (also [here](https://madnessinthedarkness.transsys.com/blog:2017:0122_black_magic_probe_bmp_on_st-link_v2_clones)) because inside is also ARM CPU reprogrammable via SWD :-) But for backing up an restoring HW01 firmware one ST-Link is enough as is. It is supported in OpenOCD out of box.

However it is a lot of work to just replace some bitmaps. Custom android app is perhaps more practical. f-droid is source of open source android apps https://f-droid.org/en/ and the gadget bridge is there https://f-droid.org/en/packages/nodomain.freeyourgadget.gadgetbridge/ however for doing modifications you must go directly to the source anyway https://github.com/Freeyourgadget/Gadgetbridge/wiki/Developer-Documentation
There are also other similar open source apps for smart watches published on f-droid that you may try to modify for HW01, gadgetbridge was first I found.

fanoush, 2019-03-07:
Oh, when googling "lenovo hw01 oled 128" it looks like it is 128x32 so same as DS-D6. Is the display also so flushed in the body so you don't see all of it when looking from side?

techieaarya, 2019-03-10:
Yes, the display is flushed in the body, can I send custom bitmap from the app? And is DS-D6 and Lenovo hw01 same? Or can we install firmware of DS-D6 onto hw01? I did not understand the Bluetooth services, but when in I gave `BT+UPGB:1` then it replied `BT+UPGB: OK` and `BT+RESET` then the screen froze with the update logo, and are you only one working on this thing?

fanoush, 2019-03-10:
>  can I send custom bitmap from the app? 

You mean show bitmap like showing text message? I don't know. I did not investigate original firmware too much. My goal was to replace it completely with something easily reprogrammable. Now with Espruino I can write my own code in javascript to do anything I want - which is mainly better watch, better alarms (more types and length of vibration, custom message) and better communication with phone. I don't care about steps or heart rate or sleep patterns. I also prefer to avoid original app as it needs lot of permissions and I don't trust it.

> And is DS-D6 and Lenovo hw01 same? Or can we install firmware of DS-D6 onto hw01? 

If not same then it is very similar, hw01 has more features like music player. I can try to modify my DS-D6 to pretend it is HW01 with old version of firmware to force it to update to latest hw01 version, can you send output of AT+VER, BT+VER and possibly also bluetooth name you see in the app? With this I also get uploadable zip of HW01 firmware update so it can be modified or used later for restoring.  This will work only if they actually published some update for HW01.  I found people are having issues with 1.0.5 updaste https://forums.lenovo.com/t5/Smartphone-Accessories-and/Lenovo-HW01-Smartband-Facing-battery-issues-after-the-Firmware/td-p/3746137 so there should be at least this one.
And if it will work after update then the hardware is the same. If not then I can still restore DS-D6 firmware via SWD.

>I did not understand the Bluetooth services, but when in I gave `BT+UPGB:1` then it replied `BT+UPGB: OK` and `BT+RESET` then the screen froze with the update logo, and are you only one working on this thing?

Normally there are BLE services like  0000190b-0000-1000-8000-00805f9b34fb used for accepting AT commands. When frozen with update logo there are different services - possibly one starting with 0000190c. for receiving the firmware update. I wanted to know how it is different from DS-D9 in this regard. It can be seen e.g. with andrid app called nRF Connect https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp&hl=en

fanoush, 2019-05-19:
Hi, just to let you know that there is now bluetooth flasher available here https://play.google.com/store/apps/details?id=com.atcnetz.ble.readwrite and also looks like HW01 firmware is available via http here care.desay.com/apk_update/HW01_11.0.0_2.0.0_005.zip so if you would want to find the bitmaps in the firmware and replace it, it is possible.