fanoush/ds-d6 #3: Stuck in AdaDFU bootloader

mogenson, 2019-09-05:
Hello, I have a DS-D6 device I'm trying to update. As per the micropython instructions I flashed the [patched Desay bootloader](https://github.com/fanoush/ds-d6/blob/master/fwdump/DS-D6-desay-bootloader-noanim-fix.zip), then the [Adafruit bootloader](https://github.com/fanoush/ds-d6/blob/master/micropython/DS-D6-adafruit-bootloader-sd132v611.zip) via the [D6 Flasher Android app](https://play.google.com/store/apps/details?id=com.atcnetz.ble.readwrite).

The DS-D6 is now advertising as AdaDFU and the D6 Flasher app detects a Nordic bootloader. However, trying to flash either of the micropython packages results in a Disconnected and `OPERATION FAILED` error. I've also tried flashing a micropython package with the nRF Connect app (since it sees the DFU service) with the same disconnect behavior.

When connecting the DS-D6 USB port to a serial cable I don't see any communication at 115200 bps or 38400 bps. The `adafruit-nrfutil` errors out with `Timed out waiting for acknowledgement from device`. I've tried swapping around the TX/RX lines in case I had them backwards.

Any tips for flashing the uncommunicative device?

fanoush, 2019-09-09:
Sorry, noticed only now. Adafruit bootloader is notlistening for both transports like Desay one, if it advertising as AdaDFU it wont work on serial at the same time. However it waits 3 seconds fo serial  DFU after each reboot or poweron on charger from empty battery. So that is the only chance to use serial (or when forced into serial mode from micropython). This is very strange. I would suggest to use different phone. Maybe also test downloaded zip with 7zip or winzip if it is really valid zip file,maybe you downloaded github webpage instead of raw zip?

fanoush, 2019-09-13:
Hi, did you resolve the issue? What was the problem?

mogenson, 2019-09-14:
Hi. I tried again today. Still cannot get any chatter on serial and AdaDFU device disconnects when attempting to do a DFU update from the D6 Flasher Android app. This is with either of the micropython zip archives from the repo. I've checked that the downloaded files are valid zip archives.

How do you restart the DS-D6? In the AdaDFU mode, the screen is off, so it's hard to tell what state it is in. If I open the device up, is there an easy to reach reset pin or pad?

fanoush, 2019-09-16:
As explained unfortunately you have no chance with serial if it is advertising over bluetooth, It is serial or bluetooth, not both. Could be you have some interference like too many wireless devices (wi-fi, bluetooth) around like mouse, headphones. In general it should work so I'd try go to different place, turn buetooth off/on on the phone, change between nrfconnect and D6 flasher (both should work)
And BTW there is also restore package that you can try https://github.com/fanoush/ds-d6/blob/master/micropython/DS-D6-adafruit-back-to-desay-sd132v201.zip 

As for reset, yes there is RST pin under OLED so you need to lift it, easier way is to momentarily shorcut battery pins on the side near usb connector (marked +,- on [this photo](https://fccid.io/png.php?id=3414019&page=2) ), that will power off the device. Then you could prepare adafruit-nrfutil command line, slide it into usb socket with usb to serial and power and it will turn on and you have up to 3 seconds to press enter for serial flashing.

Also if you have it open you can always restore it fully over SWD pins via openocd.

If you still have issues/questions you can also join the lobby as linked in toplevel readme.

coelner, 2019-09-18:
Same problem here. Both apps don't work. The bootloader disconnects the flasher.

fanoush, 2019-09-19:
Hello, I have tried this again from scratch with two devices and can confirm this is really  not working now. I am no sure what is wrong or what changed. I tried D6Flasher, nrfconnect and nrf toolbox on two phones and it is hit and miss. I did succeed once with nrf toolbox and flashed micropython blerepl to one device but I suppose I was very lucky as I could't repeat it again.

This is very strange as I don't remember having any issues when I tried this months before with my first DS-D6. I even tried it like week ago with and flashed another DS-D6  succesfully not noticing any issues like this. I guess that after first flashing I am mostly using dfu over serial but I had to flash it over bluetooth at least initially. Will investigate more.

I guess the only way out now is to let the battery die and try the serial mode which is available for few seconds when it powers on. Or open the device of course and restore via SWD. Sorry about that. Will investigate more.

fanoush, 2019-09-20:
I may have a solution. There is similar/same issue reported here `https://github.com/adafruit/Adafruit_nRF52_Bootloader/issues/35`
there is some HCI buffer inside bootloader set to 8, looks like this overflows with default nrfConnect settings of sending 10 packets at once, please go to nrfConnect settings -> DFU options and reduce Number of packets from 10 to 5 (or try even lower if 5 doesn't help). This helped me and I flashed micropython on first try.

coelner, 2019-09-26:
so far this worked for me but I flashed the uart version. 
Next problem, I can not get any uart reply from the usb pins (I used a ch340G TTL). Do you have a guide how to use a st-link v2 to reflash over jtag?

fanoush, 2019-09-26:
oh sorry 80mAh is more like 80 hours maximum, not 8 but I guess the draw is more than 1 mA, see e.g. https://devzone.nordicsemi.com/f/nordic-q-a/23444/power-consumption-on-nrf52832-with-uart so it can be like 2 days from full battery

coelner, 2019-09-27:
Thanks for your help but I think the device is dead. It does not respond to anything of this.

fanoush, 2019-09-27:
What you mean by "anything of this" - serial or SWD? Most probably you are just doing/connecting something wrong, it is quite unlikely to kill it just by flashing.   For serial connection - do you attach also 5V (not 3.3V) to usb power pin and GND in addition to RX/TX? It won't start otherwise if battery is empty. OTOH if you think it is charged but not responding then I'd just put it away for few days to let the battery discharge and start again later. Hopefully you got more than one.

The SWD should definitely work in any state of CPU/ram/flash if you attach GND,5V and SWD pins. Also you may measure 3.3 (VDD)  voltage pin to ground, next to oled, see https://fccid.io/png.php?id=3414019&page=2 to check if it has power.

Anyway if you wish you may describe exactly what you are doing in what order to check if there is some obvious mistake.

coelner, 2019-09-27:
I tried again with a ch340g and ft2322h but no uart response. Switched TX and RX and applied 5V to the USB connector. 
I tried also a st-link v2 with GND and DIO, second try was with CLK. I get no response at all, only the st-link is recognized. (It is similiar to some of the st-link clones when I try to reflash them. they do not respond to anything)
The battery is between 3.8 and 4V, it should be enough. 

Usually the device should be rebooted if some of the rst pads is connected to ground?

coelner, 2019-09-27:
I tried again with a ch340g and ft2322h but no uart response.
I tried also a st-link v2 with gnd and dio, second run was with clk. 
The battery is between 3.8 and 4V, it should be enough. 
[image]
Is this right:
1 - VDD -> 3.3V
2 - DIO
3 - CLK
4 - GND
5 - CHG

fanoush, 2019-09-27:
Yes, that is right. if battery is fine then for SWD you need 3 wires GND,DIO,CLK connected to STLINK v2 pins  GND,SWDIO,SWCLK. GND is easier to connect via usb connector if you have usb female cable.

reset is RST under OLED, for reset you need to connect it to gnd momentarily. But easier is to shortcut +/- battery briefly, that will turn it off and you will need to attach 5V to usb to wake it.

coelner, 2019-10-04:
Ok, thanks. Howe ever the RST does not work with all tries I did, the shortcut between the BAT is working properly. 
But I am curious why I can't get any cpu connection over SWD...

fanoush, 2019-10-04:
Both for reset and swd you need ground pin, are you using GND from usb connector? maybe you can scratch test points a bit to get proper contact? can you make photo of your setup? These are both really basic things that should just work (reset and SWD)

coelner, 2019-10-04:
think it was a broken pogo pin. And maybe some of the foam over the RST pad was causing no reliable connection.
At least if I run `sudo openocd -d2 -f interface/stlink-v2.cfg  -f target/nrf52.cfg` I get the  'welcome' screen and the device restarts. It shows then the clock shortly and then a battery charged status page with 100% and three battery bars.
```
[11:50|user@localhost ~]$ sudo openocd -d2 -f interface/stlink-v2.cfg  -f target/nrf52.cfg
Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 2
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
adapter speed: 10000 kHz
Info : Unable to match requested speed 10000 kHz, using 4000 kHz
Info : Unable to match requested speed 10000 kHz, using 4000 kHz
Info : clock speed 4000 kHz
Info : STLINK v2 JTAG v29 API v2 SWIM v7 VID 0x0483 PID 0x3748
Info : using stlink api v2
Info : Target voltage: 3.196342
Error: init mode failed (unable to connect to the target)
in procedure 'init' 
in procedure 'ocd_bouncer'

```

fanoush, 2019-10-06:
Oh, interesting. Just recently I noticed I have two types of stlink V2 dongles with switched pins - one is marked SWDIO,GND,SWCLK on right side and the other is SWCLK,SWDIO,GND (from top). The markings are correct on both as it is indeed switched inside but maybe you have one mixed. It is great it finally works for you.

coelner, 2019-10-19:
Did you patch this file too: https://github.com/fanoush/ds-d6/blob/master/micropython/DS-D6-adafruit-back-to-desay-sd132v201.zip

Currently I can't get into the bootloader successfully, I see only two of the three arrows.

fanoush, 2019-10-19:
yes, you are in the bootlaoder if you see the arrows :-)
it is actually the older bootloader with animations but it is fixed so softdevice can be upgraded

coelner, 2019-10-20:
Ok, something is fishy here... I can't upload and the bootloader reaches always the timeout. I tried openocd but I fail early:
```
sudo openocd -d2 -f interface/stlink-v2.cfg -f target/nrf52.cfg
Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 2
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
adapter speed: 10000 kHz
Info : Unable to match requested speed 10000 kHz, using 4000 kHz
Info : Unable to match requested speed 10000 kHz, using 4000 kHz
Info : clock speed 4000 kHz
Info : STLINK v2 JTAG v29 API v2 SWIM v7 VID 0x0483 PID 0x3748
Info : using stlink api v2
Info : Target voltage: 3.196342
Info : nrf52.cpu: hardware has 6 breakpoints, 4 watchpoints

```
telnet:
```
telnet localhost 4444
Trying ::1...
Connection failed: Verbindungsaufbau abgelehnt
Trying 127.0.0.1...
Connected to localhost.
Escape character is '^]'.
Open On-Chip Debugger
nrf52.cpu: hardware has 6 breakpoints, 4 watchpoints
> reset halt
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x000008e4 msp: 0x20000400
> flash banks
> flash
flash
  flash bank bank_id driver_name base_address size_bytes chip_width_bytes
            bus_width_bytes target [driver_options ...]
  flash banks
  flash init
  flash list
> flash list
> Connection closed by foreign host.
```
I miss something like a flash or something to write to. Any hints?

vendelin8, 2022-02-09:
When I flashed `DS-D6-desay-bootloader-noanim-fix.zip` and `DS-D6-adafruit-bootloader-sd132v611.zip`, I was able to come back with `DS-D6-adafruit-back-to-desay-sd132v201.zip`. Now I flashed `DS-D6-micropython1.10-blerepl.zip` too, and it seems to be stuck. It shows up as a `mpus` device in nrf Toolbox and D6Flasher. I sent
```
import machine
machine.enter_serial_dfu()
```
with nrfToolbox UART, nothing seemed to happen. If I try to flash `DS-D6-adafruit-back-to-desay-sd132v201.zip`.
The `Start Bootloader` button in D6Flasher writes the text about the 3 arrows, but my DS-D6 shows nothing on the screen.
I get `dfu characteristics not found` with both tools.
I googled it, the first couple of results doesn't seem to help. What do you think?

fanoush, 2022-02-09:
`enter_serial_dfu()` is not bluetooth, it is serial uart on middle usb pins. so you need usb to serial ttl adapter

see https://github.com/fanoush/ds-d6/tree/master/micropython#how-to-use for , for bluetooth you needed the first one

vendelin8, 2022-02-09:
Edited, I meant ota, sorry. I tried with that, nothing seemed to happen.

fanoush, 2022-02-09:
ok, if you still see mpus device (e.g. in nrfConnect Android app when scanning) then micropython is still running. Can you still connect to it in nrfConnect and see nordic uart service?

Not sure what  nrfToolbox UART does (e.g. regarding sending new lines), can you try https://aykevl.nl/apps/nus/ from Chrome on PC or Android phone to type in the machine.enter_ota_dfu() code?

Or there is also https://play.google.com/store/apps/details?id=de.kai_morich.serial_bluetooth_terminal&hl=en&gl=US

D6Flasher cannot handle switch from micropython (it would need to send the import+enter_ota_dfu() lines too)

vendelin8, 2022-02-09:
Ok, it worked from Android Chrome. Before I tried on desktop linux latest Chromium, but it said bluetooth not supported. From Android Brave it was the same. Now I'm back, thank you very much.