fanoush/ds-d6 #4: Exposed and not exposed pins on DS-D6?

grisevg, 2019-09-10:
Anyone knows which pins the HR sensor is on, on DS-D6? I want to hook something else to them, don't really need HR. Other than that there are only 2 gpio pins available - tx and rx? I guess if I'd manage to disconnect the HR sensors, I could also steal pin 26 if it's exposed?  any  other ones are exposed (without the anxiety smd soldering to ic pins) ? ADC?

fanoush, 2019-09-10:
Hi, pins are described in https://github.com/fanoush/ds-d6/wiki/Hardware .
See also pictute https://fccid.io/png.php?id=3414019&page=2 , HR pins are indeed easily accessible next to and under OLED. Also reset pin is pin21 and can be used as generic gpio.

BTW, for generic discussion like this you can  join  https://gitter.im/nRF51822-Arduino-Mbed-smart-watch/Lobby most of DS-D6 related talk is there.

grisevg, 2019-09-10:
Nice, I did go through the hardware page, just couldn't tell which pins were i2c hr, thank you very much for explaining and info about pin21! I'll join the gitter, sorry for making an issue for a question.

closed 2019-09-10 by grisevg

fanoush, 2019-09-10:
No problem. As for HR pins (7,8,26) I guess it is those named HSCL,HSDA,HEN on the FCCID board photo. Also maybe INT2 may be related. INT1 should be for KX023 but I am not sure about INT2 - can be KX023 or PAH8001.