fanoush/ds-d6 #5: HRS3300 getting HRS data

corteXnzt, 2019-09-29:
Did you get the HRS3300 working? I'm using c/c++ and had access to the register but i get too high values on HRS register like around 14k to 15k. At first I taught that the bpm value should be the one stored on that register because it has HRS engine already but not.  If you have a code that I can base my code then that would be a great help. TIA.

fanoush, 2019-09-30:
@atc1441 has good example here
https://github.com/atc1441/D6-arduino-nRF5/blob/master/libraries/D6Examples/examples/HeartRateTest/HeartRateTest.ino#L428

There are two parts of the problem:
1. read raw values from the sensors - that is working  and we have pretty good knowledge which registers from datasheet are needed.

2. process raw data into heart rate - for this there is closed source static library for cortex M4 - it is not freely available but is provided by manufacturer free of charge (after signing NDA?) This library can be found in some github projects near the pxialg.h header file and works fine but it is not clear what is exact status of this library. However as I understand it with knowledge mentioned in 1. it is possible to have good approximation even without this library.

fanoush, 2019-09-30:
Oh, just noticed that you talk about HRS3300, not PAH8001, sorry.

fanoush, 2019-09-30:
search github for hrs3000 (not hrs3300) and you will find the project with example, there is also keil library there

corteXnzt, 2019-10-01:
Hello fanoush, yeah PAH8001 has already an engine built in so you just fetch the data.  But with this HRS3300 only the raw values are stored on register. I misunderstood the datasheet or Must be lost from translation. 

Anyway did search last night here in github and not showing any results on mobile. Searched again this morning on PC and found what you are pointing. Actually I'm using the arduino platform based so I need to port this to work with Arduino. Thank you very much fanoush! I'll start working on this. Closing this issue (not actually but a question).

fanoush, 2019-10-01:
OK, if you get something working, please don't hesitate to share. BTW which device you have? 

I will keep this issue open for future progress since HRS3300 is not working yet and people searching for it later will see it and will not create another issue.

corteXnzt, 2019-10-01:
I'm using a B57 Smart bracelet
https://www.smartwatchspecifications.com/Device/b57-smartwatch/

I can access all parts including the accelerometer raw values at this moment.  With the HRS3300 I had written already a library to access enable and disable it and read the HRS and ALS register. 

I spent the whole morning till this afternoon exploring the code but unfortunately based on my findings the source code in the kiel has dependencies that is not included on the whole project. I first started understanding the code and proceed with porting it with arduino but when I can't trace some extern functions I moved to SES to compile the whole project but I found out that it really has missing parts.

Based on the functions on that library it sends the HRS and ALS data to a function that calculates the BP and heart rate which is declared as extern function. Tried searching that function but I guess that is already proprietary as  I can't find anything. I think their function is good even though I don't know yet what inside of it since they used the ALS in that function in calculating the BP and HR and there is also a function which take account the accel values in calculation.

My plan now is adapting my i2c code to the heart rate sensors that are read by analog which is common to manufacturers like adafruit and dfrobot. 
readAnalog>>> value>>>calculate BPM
to
i2c read analog values written on register>>>value>>>calculate BPM

Or if you have other suggestions?  Thanks again for the help.

atc1441, 2019-10-01:
Hi please share your current code with me i want to take a look at it.

I got some devices with this heartrate sensor

atc1441, 2019-10-01:
Please try to send the raw data to serial and use the serial Plotter to watch how it looks

corteXnzt, 2019-10-01:
I have closed the device so I have no access to the serial UART port currently. Im debugging  thru the ble UART. I'll try later this night with larger screen on ipad since I can't visualize it over the phone since the screen is too small.

Pardon the way this library is written since this is the first library I have made. I welcome corrections and suggestions.  I uploaded it a moment ago.

https://github.com/corteXnzt/hrs3300_test

corteXnzt, 2019-10-01:
Here are some screenshots at 3ms interval. Lower than that I get some packet loss. Some looks like a pulse signal to me. 
[image]
[image]

atc1441, 2019-10-01:
you could try to only send every 10th messaurement and then it should look better, or you could also take an averrage of the last 10 messauerements and only send that. just to see how it looks.

atc1441, 2019-10-01:
by the way, what happens if you change the width ?

atc1441, 2019-10-01:
First tests looking good

[image]

atc1441, 2019-10-01:
I made a basic HRS3300 library for Arduino/nRF52 please look here: https://github.com/atc1441/HRS3300-Arduino-Library.

you can replace the serial with bleserial and it should print the Heartrate to ble.

fanoush, 2019-10-01:
Wow, very nice, thanks. So fortunately just the __hardfp_sqrt definition fix for Keil library was needed like with pah8001 library.

atc1441, 2019-10-01:
Not exactly. This time it was __hardfp_sqrt and not __hardfp_sqrtf so not for float, but yes kind of the same.

Also the hrs does not have a seperate define file for the library  it is all within some other functions mixed up.

corteXnzt, 2019-10-02:
So that what's missing with that library. Tested your library just now atc1441. This is good, haven't compared the BP yet with another smart band but by quick test I can read from 60-83 bpm. Thank you very much for immediate testing and library! 

Question:
How did you edit the compiled lib? And I tried adding that on the segger embedded studio just in case I want to move away from arduino later on, now one of the error is >>undefined..\..\..\HRM\hrs3300_alg_dynamic.c:(i.Hrs3300_alg_send_data+0x5a): undefined reference to `__hardfp_sqrt' 
which pertains to what you two are discussing.

atc1441, 2019-10-02:
No problem, i wanted to do it anyway, and now was the right time^^

The timing is quite importand, so you need to pull at exactly 40ms or another value, but everytime the same persiod.

I did not edit the compiled library, but it is possible to do it. extract the o file change the needed hex values in binary and combine it again with gcc ar or what it name was.

for the sqrt you need to add this somewhere: https://github.com/atc1441/HRS3300-Arduino-Library/blob/master/src/HRS3300lib.cpp#L4

fanoush, 2019-10-02:
one could also add this one line __hardfp_sqrt method directly to the library, but it doesn't matter much as it is in the source that comes with it

corteXnzt, 2019-10-02:
> The timing is quite importand, so you need to pull at exactly 40ms or another value, but everytime the same persiod.
Ok noted on this. 

I added that line and the error is gone. Thanks again to both of you. Cheers!

dariosalvi78, 2020-01-02:
Hello! Any plan to include this into the Espruino-compatible bundle?