fanoush/ds-d6 #7: Building micropython for DS-D6

daniel-thompson, 2020-01-13:
Hello

Thanks for posting about all your work in DS-D6. Very inspiring!

I've been trying to replicate your Micropython build (I hope to come up with a hybrid watchdog system to implement very long press resets in a reasonably robust way).

The bootloader is working and I believe I have the right sources (I used the SHA-1 hash from os.uname() then applied your patch) but my version does not work. I'd like to narrow down a few variables, could you share what toolset you use and how you convert from hex file to DFU zip file?

fanoush, 2020-01-13:
Hello, what 'my version does not work' mean exactly?

I am building in ports/nrf via
```
make BOARD=dsd6 SD=s132
```
Then I am using adafruit-nrfutil to make the package from `build-dsd6-s132/firmware.hex` as per https://github.com/adafruit/Adafruit_nRF52_nrfutil#usage  -  the --dev-type 0x0052 is important

fanoush, 2020-01-13:
also if you search https://gitter.im/nRF51822-Arduino-Mbed-smart-watch/Lobby for "micropython" you find lot of messages mainly from @sebi5361 , he tried to get micropython working as he was quite familiar with it on other boards  but later he moved to espruino to as the nrf52 port was not so mature when compared to espruino or other micropython port

daniel-thompson, 2020-01-14:
Sorry "my version does not work" isn't very descriptive now I come to think of it. I'm trying to reproduce the BLE REPL build and "does not work" means I see no BLE advertisement when I program my build (but I do when I use your build).

I did the test multiple times to make sure there was not a flashing problem.

Yours -> worked (mpus is advertised, machine.enter_ota_dfu() to reprogram)
Mine -> fail (no advertisement, force bootloader reentry)
Yours -> worked
Mine -> fail

I can't see anything different in your genpkg approach (I also picked up the command line from the bootloader readme). I guess I'll have to fire up a debugger...

daniel-thompson, 2020-01-14:
I have been wasting your time!

The problem was *very* obvious when I looked at the backtrace!

Turns out I carefully applied your patch to micropython v1.10 and enabled BLE. When I tried to build that there were a few build problems (mostly S132 6.0.0 instead of S132 6.1.1) so I fetch the git rev from your build.

I then adopted the exact git rev, reapplied the patch, built it and tried to download it (discovering the --dev-type=0x0052 that I initial overlooked).

You might notice the re-enabling BLE was missing from the above sentence...

closed 2020-01-14 by daniel-thompson