fanoush/ds-d6 #8: Flash Micropython using Openocd ?

gerardwr, 2020-01-31:
Hi,

Have been following this github and gitter about alternative firmware for NRF52832, excellent info @fanoush !

Now I would like to flash my Nrf52832 (actually a E73.....) with Micropython using Openocd, but after much reading and many attempts I did not succeed.

Flashing Espruino .hex firmware for NRF52832 using Openocd works fine. No Openocd errors, and I can use the Espruino Web IDE on Google Chrome. The command I use on RPi is:
```
pi@raspberrypi3:~ $ sudo openocd -f interface/raspberrypi2-native.cfg -c "transport select swd" -f target/nrf52.cfg -d2 -c init -c "reset init" -c halt -c "nrf5 mass_erase" -c "program espruino.hex verify" -c reset -c exit
```

For Micropython I tried to flash 2 files from this github:
- sd_bl.bin from https://github.com/fanoush/ds-d6/blob/master/micropython/DS-D6-adafruit-bootloader-sd132v611.zip
- and the  firmware.bin from https://github.com/fanoush/ds-d6/blob/master/micropython/DS-D6-micropython1.10-blerepl.zip

But after flashing and verifying OK, the NRF does not start and Openocd reports:
```
Polling target nrf52.cpu failed, trying to reexamine
```

Below the openocd command and it's output.
```
pi@raspberrypi3:~ $ sudo openocd -f interface/raspberrypi2-native.cfg -c "transport select swd" -f target/nrf52.cfg -d2 -c init -c "reset init" -c halt -c "nrf5 mass_erase" -c "program sd_bl.bin verify" -c "program firmware.bin verify" -c reset -c exit
Open On-Chip Debugger 0.10.0+dev-00985-g2dc88e14 (2019-12-27-19:04)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 2

BCM2835 GPIO nums: swclk = 11, swdio = 25

swd
Info : BCM2835 GPIO JTAG/SWD bitbang driver
Info : JTAG and SWD modes enabled
Info : clock speed 1001 kHz
Info : SWD DPIDR 0x2ba01477
Info : nrf52.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : Listening on port 3333 for gdb connections
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x000008e4 msp: 0x20000400
Info : nRF52832-QFAA(build code: E0) 512kB Flash, 64kB RAM
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0xfffffffe msp: 0xfffffffc
** Programming Started **
Warn : Adding extra erase range, 0x0001fb38 .. 0x0001ffff
** Programming Finished **
** Verify Started **
** Verified OK **
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x0001b090 msp: 0x20001188
** Programming Started **
Warn : Adding extra erase range, 0x00025d3c .. 0x00025fff
** Programming Finished **
** Verify Started **
** Verified OK **
Error: nrf52.cpu -- clearing lockup after double fault
target halted due to debug-request, current mode: Handler HardFault
xPSR: 0x01000003 pc: 0x0002e778 msp: 0x2000ffe0
Polling target nrf52.cpu failed, trying to reexamine
Info : nrf52.cpu: hardware has 6 breakpoints, 4 watchpoints
pi@raspberrypi3:~ $ 
```

I assume my error is in flashing the 2 .bin files in the wrong memory position. Or these bins may not be suitable for flashing directly? Googles a LOT bit did nof find a solution.

Could anybody point me in the right direction, would be appreciated!

Thx.

fanoush, 2020-01-31:
Yes, indeed, they are not suitable for flashing via openocd directly. While it may work in specific cases if you know what you are doing it may not work in other cases even if you got right files and flashed it to correct address. And BTW you did not post your openocd commands, only output so it is hard to guess what exactly you did.

Anyway in this case there is no chance as sd_bl is combined softdevice and bootloader that needs to be splitted and each part belongs to different location.

If you knew what to do you could flash S132 6.1 softdevice alone (hex file from nordic), then bootloader from this zip https://github.com/fanoush/ds-d6/blob/master/micropython/DS-D6-adafruit-bootloader-long-packets.zip (belongs to 0x78000 address) and then you could hopefuly boot it and it may stay in bootloader so you can flash application zip over bluetooth.

nrf52 boot flow is like 
- softdevice boots, check if there is bootloader address in UICR, if yes then jump to bootloader, if no jump directly to application
- if bootloader starts it checks its stored bootloader settings to see if there is valid application flag  and optional crc checksum matches - if yes it jumps into application, if no it stays in DFU mode

there is lot of ways to break this flow if you flash bin files randomly

fanoush, 2020-01-31:
Oh, now I see you used all the commands on single openocd command line, interesting. Anyway, program command with bin file without specific address has zero chance of success too. hex file has correct location stored inside, bin files have no such info, it is just raw data.

gerardwr, 2020-01-31:
@fanoush Thank you for your extensive and clear response.

I have been using a single openocd command, that makes it a lot easier to try the (MANY) falshes I tried ;-)

The openocd command I used was in the output above, but it is looong. It reads:
**_sudo openocd -f interface/raspberrypi2-native.cfg -c "transport select swd" -f target/nrf52.cfg -d2 -c init -c "reset init" -c halt -c "nrf5 mass_erase" -c "program sd_bl.bin verify" -c "program firmware.bin verify" -c reset -c exit_**

OK, the .bin files can not be used in the way I hoped:
- The SD and BL part of the sd_bl.bin are targeted at different areas in the NRF memory map.
- the firmware.bin should be flashed at address 0x78000. In my command it is flashed probably at address 0x0 and overwrites the softdevice.

Just for fun I will try your suggestion to:
- flash softdevice .hex
- flash dsd6_nrf52832_bootloader-0.2.11-3-gfb847b9-dirty-nosd.bin at address 0x78000
- flash firmware.bin using BT

Your description of "nrf boot flow" is very informing and will help me in my quest.

Appreciate your help!

gerardwr, 2020-01-31:
Following your suggestion I tried:
**_pi@raspberrypi3:~ $ sudo openocd -f interface/raspberrypi2-native.cfg -c "transport select swd" -f target/nrf52.cfg -d2 -c init -c "reset init" -c halt -c "nrf5 mass_erase" -c "program s132_nrf52_6.1.1_softdevice.hex verify" -c "program dsd6_nrf52832_bootloader-0.2.11-3-gfb847b9-dirty-nosd.bin verify exit 0x78000" -c reset -c exit_**

Flash runs without errors, but after reboot no BLE device is found with Nordic NRFconnect App.

I will forget about Micropython for now, and focus om Espruino. That flashes and runs fine.

My ultimate goal is to flash my new F18 smartwatch with Espruino or Micropython, but NOT before I have some experience with the NRF52832 module.

Thanks for your time.

Output from the Openocd command was:
```
pi@raspberrypi3:~ $ sudo openocd -f interface/raspberrypi2-native.cfg -c "transport select swd" -f target/nrf52.cfg -d2 -c init -c "reset init" -c halt -c "nrf5 mass_erase" -c "program s132_nrf52_6.1.1_softdevice.hex verify" -c "program dsd6_nrf52832_bootloader-0.2.11-3-gfb847b9-dirty-nosd.bin verify exit 0x78000" -c reset -c exit
Open On-Chip Debugger 0.10.0+dev-00985-g2dc88e14 (2019-12-27-19:04)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 2

BCM2835 GPIO nums: swclk = 11, swdio = 25

swd
Info : BCM2835 GPIO JTAG/SWD bitbang driver
Info : JTAG and SWD modes enabled
Info : clock speed 1001 kHz
Info : SWD DPIDR 0x2ba01477
Info : nrf52.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : Listening on port 3333 for gdb connections
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x00000a80 msp: 0x20000400
Info : nRF52832-QFAA(build code: E0) 512kB Flash, 64kB RAM
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0xfffffffe msp: 0xfffffffc
** Programming Started **
Info : Padding image section 0 at 0x00000b00 with 1280 bytes
Warn : Adding extra erase range, 0x00025150 .. 0x00025fff
** Programming Finished **
** Verify Started **
** Verified OK **
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x00000a80 msp: 0x20000400
** Programming Started **
Warn : Adding extra erase range, 0x0007bc24 .. 0x0007bfff
** Programming Finished **
** Verify Started **
** Verified OK **
shutdown command invoked
pi@raspberrypi3:~ $

fanoush, 2020-02-01:
Since you were doing mass erase the UICR space gets cleared too so there is no pointer to bootloader and softdevice jumps directly to (missing) application. So if running mass erase (which is not needed, unless you really want to clear uicr) you need additional hex or bin file to flash to bank 1 (flash write_bank 1 file as per [openocd help](http://openocd.org/doc/html/Flash-Commands.html)), e.g. this one https://github.com/fanoush/ds-d6/blob/master/fwdump/dsd6-uicr.bin

gerardwr, 2020-02-01:
@fanoush 

Making progress with your remark on UICR ;-)

I had previously seen the nrf52.uicr bank, I already wondered where the UICS bank went ;-)

I used mass-erase because my E73 module was locked, and thought this as an extra measure to get rid of the original firmware in te module. Got rid of nrf52.uicr too.

Excellent, getting some progress now. I extended my previous Openocd command with flashing dsd6-uicr.bin to bank 1:

**_pi@raspberrypi3:~ $ sudo openocd -f interface/raspberrypi2-native.cfg -c "transport select swd" -f target/nrf52.cfg -d2 -c init -c "reset init" -c halt -c "nrf5 mass_erase" -c "program s132_nrf52_6.1.1_softdevice.hex verify" -c "program dsd6_nrf52832_bootloader-0.2.11-3-gfb847b9-dirty-nosd.bin verify 0x78000" -c "flash write_bank 1 dsd6-uicr.bin" -c "flash banks" -c reset -c exit_**

Result:
- flash progress runs without errors or warnings (see opencd log.txt below)
- in the NrfConnect App my device now appears named "AdaDFU" (see screen dump below)

Next step is trying to flash an application using DFU.

Many thanks for your help so far.  Will keep you posted.

[openocd log.txt](https://github.com/fanoush/ds-d6/files/4142940/openocd.log.txt)

[image]

gerardwr, 2020-02-01:
Success!!

Using the NRFConnect App I uploaded the file https://github.com/fanoush/ds-d6/blob/master/micropython/DS-D6-micropython1.10-blerepl.zip to my E73 module:

NOTE : The upload was stuck at 0% at first, but after reducing the NrfConnect number of packets from 10 to 5 as you described here it worked OK.
https://github.com/fanoush/ds-d6/issues/3#issuecomment-533422347

When the upload was finished I was able to connect to the WEBREPL, using the CHROME browser and URL https://aykevl.nl/apps/nus/

Connect to the Nrf device called "mpus":

[image]

My first Micropython statement executed :-)

[image]

This is my Development Platform from https://www.aliexpress.com/item/4000242009052.html?spm=a2g0s.9042311.0.0.585f4c4dtAjTQm

[image]

Many thanks for your help @fanoush, would have been impossible without it!

AkosLukacs, 2020-02-02:
Re your Development Platform: Ebyte sells a "devboard" with pins broken out & USB-UART: https://www.aliexpress.com/item/32968640384.html?spm=2114.12010615.8148356.6.5e2d37defrP6lh
If you are interested in longer range with external antenna, the other version of the module with PCB antenna has an IPEX connector, so if you solder the capacitor from the PCB antenna trace to the IPEX, you can use that with external antennas: https://www.aliexpress.com/item/32819293925.html?spm=2114.12010615.8148356.3.5e2d37defrP6lh

gerardwr, 2020-02-02:
@AkosLukacs 
Hi,

I have an F18 Smartwatch that I want to use for some experiments.

But before I replace the stock firmware on the F18 I wanted to experiment with the cheapest NRF52832 solution, so I bought this:
https://nl.aliexpress.com/item/4000242009052.html?spm=a2g0s.9042311.0.0.44554c4dq3J1BK

They have an NRF52832 board with pins and USB-UART, but I was too cheap to buy this one. BTW : the board in your link has an NRF5210 chip on it, but they have an NRF52832 too:
https://nl.aliexpress.com/item/32968664414.html?spm=a2g0o.productlist.0.0.4f777d59UIQy0z&algo_pvid=edd2470f-8ff9-4ed8-91a5-8fd3de7dc99c&algo_expid=edd2470f-8ff9-4ed8-91a5-8fd3de7dc99c-4&btsid=49c0a7d5-b09f-4c3b-9f1a-e34584cec77c&ws_ab_test=searchweb0_0,searchweb201602_7,searchweb201603_53

I have also bought this NRF52840 module. This chip has built-in USB hardware, should require only 2 resistors to connect to a USB port. And this chip is supported by the Adafruit Circuitpython firmware.
https://nl.aliexpress.com/item/32944356249.html?spm=a2g0s.9042311.0.0.d4904c4dOnlkW7

Now that I have Micropython running I realize that there is no real WebBluetooth IDE solution for Micropython. Running Espruino on NRF52 is very well supported by the "Espruino web IDE", that very usable.

I enjoy reading the communication on Gitter between you and @fanoush about NRF52832 smartwatches. It inspired me to start my experiments.

Have been using ESP8266 for 5 years now, but that's not suitable for a "wearable".

Cheers.

AkosLukacs, 2020-02-03:
My bad with the link, just clicked on the image without reading. But looks like you already knew it :)
Got the test board from "Cojxu" store last year 11.11, sold it for less than the official aliex stores (CDSENET, CDEBYTE - both were listed in Ebyte's webpage last time I checked) ever. Dunno, maybe a couple have fallen off from a truck...

[Espruino sort of works on nRF52840](http://forum.espruino.com/conversations/323658/) - you can connect via BLE, and basics work, but for example bluetooth scan fails. Hopefully it will be better because [Gordon will work on RAK5010 that has nRF52840](http://forum.espruino.com/conversations/339245/). I have that module too. Sitting in a box for now...

gerardwr, 2020-02-03:
@AkosLukacs 
Just checked the Cojxu store on AliExpress, lots of choice in modules, and good prices!

Based on my current experience with the bare modules, I think that buying an official Nordic USB dongle would have been an easier choice. Like the "Ebyte devboard" you bought, or something like this:
[image]

I read the link on the use of Espruino on NRF52840, good to know that it "sort of works" ;-)
Maybe I better stay with the NRF52832 for now, that runs Espruino quite well.

Experimenting now with the Ds-D6-Micropyton install. Use it with the WebBluetooth IDE of https://aykevl.nl/apps/nus/
Unfortunately the connection times out in 1 minute or so, and you have to reconnect again. So experiments have to short :-( 

Tried using the Espruino Web IDE for connecting to the Ds-D6-Micropyton install. WebRepl "sort of works", but also times out within 1 minute.
[image]

Any tips for a better working Micropython WebBluetooth IDE?