fanoush/ds-d6 #9: DFU file not found (espruino_2v07.13_p8_SDK11_SD20.zip)

dapgo, 2020-09-30:
Trying to flash the latest version of espruino on a P8b with espruino 2.06.129

I receive always following errors :
GATT ERROR using daflasher
DFU file not found using Nrf Toolbox and file type "distribution package (ZIP)"

the watch even  was turned off as it ran out of battery. 
From espruino web IDE, the espruino is responsive and i can flash files and the watch execute them properly.

Is possible that the package has any error?

jeffmer, 2020-10-19:
Hi, if my memory is correct, I saw the same error message in DaFlasher when I forgot to tick the Use Nordic Bootloader box.

dapgo, 2020-10-20:
During the many unsuccessful attempts to flash I try different combination of these checkboxes, so in my case was not the problem. Also I was trying with daflasher and nrftoolbox.

Fortunately, I succeed  flashing espruino 7v7.30  on this p8 watch using nrftoolbox from a different phone (android 7 and totally clean on bluetooth apps)

I don't know the root cause of the problem, maybe radio interference,  incompatibilities other android apps (tough i was hibernating gadgetbridge and dafit), the android version, ... 

My recommendation is that as soon as you find problems, try with a different phone and even turn off most bluetooth devices in the area.