fanoush/ds-d6 #10: colmi p8 bootloader bricked

joerg555, 2020-11-26:
I have done the followig:
espruino-xxx-DaFit.bin bin file upload via DaFlasher
it works, with Espruino IDE, but no mem, ...

because DaFlasher cold not do any more, i flashed with espurino with tis script:
P8 espruino bootoader flashing example 

But now the P8 is bricked ?

I habe to open it, and connect to serial or is ther any other chance to unbrick.

Please update your Readme with this issue

fanoush, 2020-11-26:
You probably mean this https://gist.github.com/fanoush/c65d0de750a87262fcdd1d91d2cdd43d
it worked for me, that is how I upgraded my P8, however you don't describe what you did exactly in which steps so I don't know when it went wrong.

There are more details in wiki
https://github.com/fanoush/ds-d6/wiki/Replacing-Nordic-DFU-bootloader
or for other devices F07 or DK08
https://github.com/fanoush/ds-d6/tree/master/espruino/DFU/F07
https://github.com/fanoush/ds-d6/tree/master/espruino/DFU/DK08
did you follow some steps from those ones?

At what point it was bricked? you used that espruino code to flash bootloader and then what? What you did next? You rebooted vie E.reboot() and did not get Espruino or ATCDFU device visible over bluetooth?

fanoush, 2020-11-26:
Oh, maybe I know, you got ATCDFU but then tried downgrade to SDK11 bootloader via https://github.com/atc1441/DaFlasherFiles#fitbootloaderdfu201zip?

But most probably you did not fix UICR bootloader settings at 0x10001018 ? That would indeed brick it while downgrading to SDK11 or SDK12.  It would work when uploading Espruino zip for SDK14, that does not need UICR to be fixed.

Anyway the P8 readme is very brief as it was the first and the step 'option 1' https://github.com/fanoush/ds-d6/tree/master/espruino/DFU/P8#espruino  just links the gist (marked TODO WIP) without any explanation what is needed. I should fix that and link the bootloader replacement wiki.

fanoush, 2020-11-26:
I mad a chage here https://github.com/fanoush/ds-d6/commit/27a80f0b873064843c1e4a2aff33e59e04eb6b22 to explain more.

If you really bricked you watch by trying to downgrade bootloader/softdevice you unfortunately need to open the device and use SWD to restore it like described here https://wiki.pine64.org/index.php/Reprogramming_the_PineTime

joerg555, 2020-11-27:
Hi fanoush
Thanks for your quick answer.
I did the following:

- flashed "espruino_2v05.438_p8-dafit.bin" with DaFlasher

- try to flash "DaFitBootloader23Hacked.bin" with da-flasher, but id don't work

- Start Espruino IDE and connect

- load https://gist.github.com/fanoush/c65d0de750a87262fcdd1d91d2cdd43d and transfare to P8

after this i see the flash message, and no answer
i wait for 10 minutes, and then disconnect :)

So my mistake was not to "Fix UICR at 0x10001018 to point to 0x7e000 so that future bootloader or softdevice DFU update works" before ?
Or is there a chance to wait 5 days if battery is low and have a hard reboot ?

It will be nice to have a backup script for orginal P8 Firmware, that was my opinion with this aktion.

So now i have to open P8 and get an "ST-Link V2" from ebay.

I have read in the wasp-os dokumentation that the new P8 has "unsupported step counter module"
https://wasp-os.readthedocs.io/en/latest/install.html#colmi-p8
you know any thing about this, may be i can check the new step counter, when i open my P8

73 Jörg

fanoush, 2020-11-27:
from your description the device should not be bricked, please describe "after this I see the flash message, and no answer"
also do you still see device over bluetoth and can connect again via webide?
if yes try to press enter, maybe you just don't see echo as per message
```
echo(1);
// include end of line after echo(1) or press enter if no prompt is shown
```
or paste `echo(1);` via clipboard again (ctrl+v) and press enter.

joerg555, 2020-11-27:
Thanks,
but it's not visible in bluetooth

joerg555, 2020-12-02:
Hi fanoush,
can you help me ?
Now i have my  P8 with st-link and openocd, and i think, i have "only" to reset uicr, do you know the steps to do this ?, sorry my frist try is in Windows, but it works, i can also do it in ubuntu.
[image]

```
bin\openocd.exe --file scripts\interface\stlink.cfg -f scripts\target\nrf52.cfg
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : clock speed 1000 kHz
Info : STLINK V2J17S4 (API v2) VID:PID 0483:3748
Info : Target voltage: 3.301808
Info : nrf52.cpu: hardware has 0 breakpoints, 2 watchpoints
Info : starting gdb server for nrf52.cpu on 3333
Info : Listening on port 3333 for gdb connections

Open On-Chip Debugger
> reset init
timed out while waiting for target halted
TARGET: nrf52.cpu - Not halted
> flash banks
#0 : nrf52.flash (nrf5) at 0x00000000, size 0x00000000, buswidth 1, chipwidth 1
#1 : nrf52.uicr (nrf5) at 0x10001000, size 0x00000000, buswidth 1, chipwidth 1

> targets
    TargetName         Type       Endian TapName            State
--  ------------------ ---------- ------ ------------------ ------------
 0* nrf52.cpu          hla_target little nrf52.cpu          reset

```

fanoush, 2020-12-02:
good progress, with swd  working you can flash anything you want :-) sorry I won't walk you through it step by step as I don't know what you want  to do exactly and learning it yourself gives you the most.

Yo can dump the UICR area to see what is there, you can dump the 0x7a000 area to see if you flashed bootloader correctly or you can mass erase everything and start from scratch, depends what is easier - as for clearing only UICR you need to poke NVMC registers directly from opencd in same way as the espruino code does https://github.com/fanoush/ds-d6/wiki/Replacing-Nordic-DFU-bootloader#espruino-examples the registers are described [here](https://infocenter.nordicsemi.com/topic/com.nordic.infocenter.nrf52832.ps.v1.1/nvmc.html?cp=4_2_0_10_3#concept_p4n_mbz_vr)