fanoush/ds-d6 #11: Can a stock firmware backup be implemented?

androidonis, 2021-05-25:
Hi fanoush, first many thanks for making it possible to use the loader on MOY-TON P8b. But I feel I was to fast as my P8b sensor configuration seems to be different, and still today, I could not find a match for the heartrate sensor. And HR is an important feature (for me).
Maybe I need to know more details on this intallation of the bootloader. My understanding is, that DA Flasher makes the stock firmware believe your program is a valid update, and then your program is executed from SPI Flash or Ram. At this point the stock firmware should be still intact in the nRF Flash and a dump over Bluetooth would be possible - or am I wrong?
Later in the process, as it is good to have SWD access, the protection bit is cleared and at that point stock firmware is lost.

With so many variants claiming to be P8, it would be a good feature to have the stock flash dump. Sometimes it is benifical to revert to stock firmware and analyse I2C communication to see how stock firmware initialises the sensors.

fanoush, 2021-05-25:
> My understanding is, that DA Flasher makes the stock firmware believe your program is a valid update, and then your program is executed from SPI Flash or Ram. At this point the stock firmware should be still intact in the nRF Flash and a dump over Bluetooth would be possible - or am I wrong?

the nordic bootloder flasher bin is  'normal' dafit application upgrade so it gets flashed first which overwrites original app. so after this there in no original firmware. There is no easy way.  There are two ways - finding some exploit/bug in the stock app itself or nordic BLE stack to execute code remotely or power glitching to unlock SWD without erasing flash - both are not easy.

> Later in the process, as it is good to have SWD access, the protection bit is cleared and at that point stock firmware is lost.

No, unlocking from the app itself via software keeps everything intact, at least on 52832, however just flashing anything via dafit upgrade procedure erases previous app completely.

androidonis, 2021-05-28:
So the stock firmware branches to a bootloader after file transfer and bootloader reflashes. If thats the case, no chance for backup. Had anybody luck with this hack: 
https://limitedresults.com/2020/06/nrf52-debug-resurrection-approtect-bypass-part-2/  
It seems that a well timed spike on DEC1 pin can void approtect.

closed 2021-05-28 by androidonis