fanoush/ds-d6 #13: Espruino Build *** No rule to make target  ***

sebi5361, 2021-07-04:
Following your [instructions](https://github.com/fanoush/ds-d6/tree/master/espruino#how-to-rebuild) to build the Espruino firmware for the DS-D6 (`make` with the `NRF_SDK11=1` argument), `make` stops at **No rule to make target**: 
```
krut@laptopkrut2016:~/Espruino$ make -j BOARD=MDBT42Q RELEASE=1 NRF_SDK11=1 DFU_UPDATE_BUILD=1
CC /home/krut/Espruino/targetlibs/nrf5x_11/components/toolchain/gcc/gcc_startup_nrf52.o
Generating platform configs
Generating pin info
Generating JS wrappers
WRAPPERSOURCES = src/jswrap_array.c src/jswrap_arraybuffer.c src/jswrap_dataview.c src/jswrap_date.c src/jswrap_error.c src/jswrap_espruino.c src/jswrap_flash.c src/jswrap_functions.c src/jswrap_interactive.c src/jswrap_io.c src/jswrap_json.c src/jswrap_modules.c src/jswrap_pin.c src/jswrap_number.c src/jswrap_object.c src/jswrap_onewire.c src/jswrap_pipe.c src/jswrap_process.c src/jswrap_promise.c src/jswrap_regexp.c src/jswrap_serial.c src/jswrap_storage.c src/jswrap_spi_i2c.c src/jswrap_stream.c src/jswrap_string.c src/jswrap_waveform.c libs/compression/jswrap_heatshrink.c libs/filesystem/jswrap_fs.c libs/filesystem/jswrap_file.c libs/math/jswrap_math.c libs/graphics/jswrap_graphics.c libs/network/jswrap_net.c libs/network/http/jswrap_http.c libs/network/js/jswrap_jsnetwork.c libs/bluetooth/jswrap_bluetooth.c libs/crypto/jswrap_crypto.c libs/neopixel/jswrap_neopixel.c
Generating pin info
make: *** No rule to make target '/home/krut/Espruino/targetlibs/nrf5x_11/components/nfc/ndef/connection_handover/ble_pair_msg/nfc_ble_pair_msg.o', needed by 'espruino_2v09.129_mdbt42q.elf'.  Stop.
make: *** Waiting for unfinished jobs....
DEFINES = -DGIT_COMMIT=30a9f382 -DNO_ASSERT -DRELEASE -DBUILDNUMBER="129" -DMDBT42Q -DHAL_NFC_ENGINEERING_BC_FTPAN_WORKAROUND=1 -DCONFIG_GPIO_AS_PINRESET -DNRF_BLE_GATT_MAX_MTU_SIZE=53 -DNRF_BLE_MAX_MTU_SIZE=53 -DBLUETOOTH_NAME_PREFIX="MDBT42Q" -DNEOPIXEL_SCK_PIN=23 -DUSE_BOOTLOADER -DUSE_DEBUGGER -DUSE_TAB_COMPLETE -DUSE_HEATSHRINK -DUSE_FILESYSTEM -DUSE_MATH -DUSE_GRAPHICS -DUSE_NET -DUSE_NETWORK_JS -DBLUETOOTH -DUSE_CRYPTO -DUSE_SHA256 -DUSE_SHA512 -DUSE_NEOPIXEL -DUSE_NFC -DNFC_HAL_ENABLED=1 -DS132 -DNRF_SD_BLE_API_VERSION=2 -DBLE_STACK_SUPPORT_REQD -DSWI_DISABLE0 -DSOFTDEVICE_PRESENT -DFLOAT_ABI_HARD -DNRF52_SERIES -DNRF52 -DNRF52832_XXAA -DNRF52_PAN_74 -DNRF52832 -DNRF5X -DNRF5X_SDK_11 -DUSE_APP_CONFIG -DARM -DLINK_TIME_OPTIMISATION -DEMBEDDED
```
Without the `NRF_SDK11=1` argument, the build works fine thought. Could you help?

fanoush, 2021-07-05:
The error is related to NFC so try to disable NFC in board file

fanoush, 2021-07-05:
Then the higher MTU configuration is not suppported with softdevice 2.0 so I'd better disable that one.

sebi5361, 2021-07-05:
I commented out `NFC` as well as `#DEFINES+=-DNRF_BLE_GATT_MAX_MTU_SIZE=53 -DNRF_BLE_MAX_MTU_SIZE=53',`.
It went further but now it is stuck at:
```
CC /home/krut/Espruino/targetlibs/nrf5x_11/components/drivers_nrf/clock/nrf_drv_clock.o
targetlibs/nrf5x_11/components/drivers_nrf/twis_slave/nrf_drv_twis.c:19:2: error: #error "TWIS driver included but none of TWIS devices is activated in the configuration file"
 #error "TWIS driver included but none of TWIS devices is activated in the configuration file"
  ^
cc1: warning: unrecognized command line option '-Wno-expansion-to-defined'
make: *** [Makefile:798: /home/krut/Espruino/targetlibs/nrf5x_11/components/drivers_nrf/twis_slave/nrf_drv_twis.o] Error 1
make: *** Waiting for unfinished jobs....
cc1: warning: unrecognized command line option '-Wno-expansion-to-defined'
```
I will investigate further...
Any idea on how to solve this issue?

I read the differences [you pointed out](https://github.com/fanoush/ds-d6/wiki/DFU-update#dfu-package) between SDK 11 / SDKs > 11 and how the  bootloader is not/is secured.
If using SWD to flash the firmware is SDK 11 also mandatory?

I didn't see an external low-frequency 32768 Hz crystal on the DS-D6 [PCB](https://raw.githubusercontent.com/fanoush/ds-d6/master/photos/extra-pins.png).
Does Esprino uses the internal RC oscillator instead?

fanoush, 2021-07-05:
sdk11 is not mandatory, in fact nowadays I'd suggedst to upgrade to sdk12, see https://github.com/fanoush/ds-d6/tree/master/espruino/DFU/SDK12 , no swd needed just fixed bootloader and then upgrade package

sebi5361, 2021-07-05:
Thank you very much!
I will try that.

An additional question:
Is there a way to build Espruino enabling the external-low frequency crystal? (for a board equipped with it)

closed 2021-07-05 by sebi5361