fanoush/ds-d6 #14: bootloader-softdevice

vendelin8, 2022-02-08:
Hi,

Thanks for all your work, this is very cool.
I'm trying to build and flash to DS-D6 with Go through github.com/tinygo-org/tinygo/ . It supports softdevice v6. There's an Adafruit bootloader in your files to that has SD v6, but generally if I flash it, I get stuck, can do nothing other than coming back with `DS-D6-adafruit-back-to-desay-sd132v201.zip` as suggested here: https://github.com/fanoush/ds-d6/issues/3 . But that's SD v2, so I can't use it with Tinygo :D
So my question is: how could I create a desay bootloader-softdevice image with SD v6?

Thanks again.

fanoush, 2022-02-08:
> I get stuck, can do nothing other than coming back with `DS-D6-adafruit-back-to-desay-sd132v201.zip`

 If you can come back via that other package then it worked and you were on S132 6.1.1 with adafruit bootloader

> So my question is: how could I create a desay bootloader-softdevice image with SD v6?

maybe your  question should be   "Why am I stuck?"

closed 2022-02-08 by vendelin8

fanoush, 2022-02-08:
> so I guess my application image is not good enough.

yes, it would need to be something made by adafruit-nrfutil as described on adafruit bootloader github And built for S132 6.1.1.