fanoush/ds-d6 #15: DS-D6 for tinygo

vendelin8, 2022-02-09:
I'd like to try building for DS-D6 with tinygo, and I'd like to ask you to look at my code before I flash it. It compiles, I created a zip with adafruit-nrfutil, and I don't want to tear it apart :D But I will take the responsibility anyway. All I'd like is to please look at it, and check if you find anything wrong.

So my code looks like the following:
```
package main

import (
	"device/arm"
	"device/nrf"
	"machine"
	"time"

	"tinygo.org/x/bluetooth"
)

const vibrMS = time.Duration(5)

var (
	adapter = bluetooth.DefaultAdapter
	vibr    machine.Pin
)

func main() {
	// first I initialize bluetooth, so I can connect to it; probably it enables D6Flasher to enter bootloader mode if anything goes wrong
	must("enable BLE stack", adapter.Enable())
	adv := adapter.DefaultAdvertisement()
	must("config adv", adv.Configure(bluetooth.AdvertisementOptions{LocalName: "Vibry1"}))
	must("start adv", adv.Start())

	// the motor pin that I defined in the machine package to pin = 25
	vibr = machine.VIBRATOR_PIN
	vibr.Configure(machine.PinConfig{Mode: machine.PinOutput})
	vibrFor(20, 1000) // vibrating a little for 20 seconds, I'll try to enter bootloader mode meanwhile

	// trying to enter bootloader mode by itself
	nrf.POWER.SetGPREGRET(0xA8)
	arm.SystemReset()

	vibrFor(30, 3000) // vibrating a little for 30 seconds, I'll try to enter bootloader mode to check if we're still here
}

func vibrFor(count, ms int) {
	for i := 0; i < count; i++ {
		vibr.Low()
		time.Sleep(time.Millisecond*time.Duration(ms) - vibrMS)

		vibr.High()
		time.Sleep(time.Millisecond * vibrMS)
	}
}

func must(action string, err error) {
	if err != nil {
		panic("failed to " + action + ": " + err.Error())
	}
}
```

What do you think?

vendelin8, 2022-02-11:
I tried it, and I couldn't flash it. I always get `Operation failed`. I'm building a zip with adafruit-nrfutil. I tried many different parameter sets. At the end I tried yours that are in `DS-D6-micropython1.10-blerepl.zip`, as I was able to flash that lately. They are quite weird though. Your `--sd-req 0xFFFE` doesn't match any SD I've seen. Same for dev revision. So I ended up with `dfu genpkg --application blink1.hex --sd-req 0xFFFE --dev-type 0x52 --dev-revision 44463 blink1.zip` but it had `Operation failed` too. My manifest differed only in application bin and dat name, and crc. Same device, same BL+SD. I tried to google the error message, some thread told me to sign the package, but I didn't want to, and it should work without that. Can you please help?

vendelin8, 2022-02-12:
Thanks. I removed those, and I could flash it. And thanks for noting that there's no 32kHz crystal in it, I changed the machine code with this.
So far: bluetooth seems to work. I could reflash it. It surprised me, the SD kept the AdaDFU as a device, and another one `Vibry` showed up that I initialized from my code. Vibration works, but 5 ms was too small, 20 is noticeable. And it went to bootloader mode after 20 seconds. So far so good. Thanks for all your help.

closed 2022-02-12 by vendelin8

vendelin8, 2022-02-14:
I got it unresponsable :-\ But I'm not into the hardware part. I tried to read your guides and other sources, but I'm still very uncertain.
Basically, it has adafruit bootloader, so if the battery goes down, at charge start I have 3 seconds to serial dfu, right? What kind of cable is that, and how do I connect it to a computer? Is it USB to TTL Serial cable? That goes to another microcontroller? Eg. I have a digispark around, and some rPi like boards. Please help.

fanoush, 2022-02-14:
Yes that  happens sooner or later :-) yes tou need usb to ttl serial cable - something based on cp2104 or cp2102 or Ch341 or anything like that from ebay/aliexpress
and then use `adafruit-nrfutil dfu serial` as described in https://github.com/fanoush/ds-d6/wiki/DFU-update#serial-dfu-procedure , however with adafruit bootloader you don't change speed to 38400, you keep default 115200.

One can prevent that by using watchdog in the code and enter DFU as a first thing in the code when watchdog reset reason is detected at startup.

Also the adafruit bootloader could be modified to do that - it is quite old, today I would add few things to it to make it more brick proof. You can also try to touch button after you put it on charger but most probably it won't work as it is touch and not physical button so it takes some time to register the touch. To more recent bootloaders I have added vibration+wait 300ms for button touch when put on charger with dead battery.

BTW it is also relatively easy to open (see the trick here https://youtu.be/3gjmEdEDJ5A?t=1418 but I'd use something plastic or wooden as a lever to not scratch it) and then you need some swd adapter like stlink v2 clone  - even raspbery pi or raspberry pico or esp32 can do that over two wires with right software.

vendelin8, 2022-02-27:
I'll receive the parts soon, so just to make sure. I bought a USB A female connector that I can put between the ds-d6 and the ttl-serial chip. The two side cables of the usb are gnd and vcc. But are the middle ones tx and rx, so I don't have to open it actually, just put in the usb female, and connect all 4 wires to my ttl-serial converter. Right?

vendelin8, 2022-02-27:
After I get it back, I plan to play around with it. When I'm ready, I'll add ds-d6 to tinygo, and make a PR to you repo with some demo code too. If you're okay with that. And BTW, how do you know that it doesn't have a 32kHz crystal? As far as I've read, it's inside nrf52832, which is the core of ds-d6.

fanoush, 2022-02-28:
> just put in the usb female, and connect all 4 wires to my ttl-serial converter. Right?

yes, see https://github.com/fanoush/ds-d6/wiki/DFU-update#serial-dfu-procedure
at worst I mixed the rx/tx pins, not sure really

as for external 32kHz crystal - it is optional component and is simply not visible anywhere on the board, also I verified that softdevice initialization will fail if you select 32kHz crystal as clock source