fanoush/ds-d6 #16: Dashes in board filename lead to error in Espruino prpvision.sh

arglurgl, 2022-03-13:
Hi, I am not sure how critical this is, but when using the recommend way of building Espruino (https://github.com/espruino/Espruino/blob/master/README_Building.md) the prvision.sh bash script will throw an error. E.g. for the SMA B5

    me@mypc:~/code/Espruino$ source scripts/provision.sh B5-SDK12
    bash: export: "PROVISION_B5-SDK12=1": not a valid identifier
    Provision BOARDNAME = B5-SDK12
    Provision FAMILY = NRF52
    ===== NRF52
    ===== ARM
    arm-none-eabi-gcc installed
    me@mypc:~/code/Espruino$ 

I can't tell if this error actually breaks something, but it might be worth removing the dashes from the board files. Even if only to avoid confusion. Simply removing the dash from the filename and adapting the compile commands did the trick for me.

fanoush, 2022-03-14:
Oh, thanks for reporting. I guess I've never run provision.sh for that.  Looks like provision.sh is quite simple and does not handle parsing board name from board file which would help in this case. Like this line https://github.com/espruino/Espruino/blob/master/scripts/provision.sh#L46 the script should read also `'board.info["boardname"]'` and override BOARDNAME variable with it if present. Then it would use 'B5' instead of B5-SDK12

Anyway, the  error looks harmless and provision seem to work despite the error message.
I have quite a lot of other board files with dash in name for other devices so won't rename them all. Fixing provision.sh seems easier.

arglurgl, 2022-03-14:
OK, I'll mention this on the Espruino issues, let's see if they fix it. Else the workaround is easy enough.

fanoush, 2022-03-15:
> let's see if they fix it. 

It is better to create pull request directly if you can do it.

> OK, I'll mention this on the Espruino issues

And BTW the issue is not primarily about dashes but about determining correct board name. The idea behind boardname override in BOARDNAME.py is  that you can have more board configuration files for same board (see e.g. https://github.com/fanoush/ds-d6/tree/master/espruino/DFU/P8 ). In this case you could have RASPBERRYPI_XX.py RASPBERRYPI_YY.py with different stuff built in or LINUX_XX.py and the provision.sh should still do the stuff for LINUX or RASPBERRY board.

arglurgl, 2022-03-16:
Yeah I could not really get my head around what the script is trying to do and as far as I understood, this may break other things, see https://github.com/espruino/Espruino/issues/2172 . But as there is a hint now on the docs of Espruino (see linked issue) I think its fine as-is so I'll close this. And regarding PRs I might pester you with one as soon as I get dual storage working for B5 in Espruino 2v12 (https://github.com/arglurgl/smartwatch-sma-b5-espruino/tree/main/espruinobuild) but that's a bit off-topic here. ;)

closed 2022-03-16 by arglurgl

fanoush, 2022-03-17:
oh, well, creating issue about dashes was not the way to get it fixed as I explained