fanoush/ds-d6 #17: golang watchdog

vendelin8, 2022-06-06:
Hi,

I got my serial to ttl, and was able to flash my ds-d6 through serial. Now I'm trying to add a watchdog, reading all related stuff, not really working. Maybe you can help.

I don't have a debugger, so I'm trying to get info of vibrations, but it's hard when it breaks :)

I have something like this now:

```
... main ...
	if nrf.POWER.GetRESETREAS_DOG() == 1 { // reset from watchdog
		nrf.POWER.SetGPREGRET(nrf.POWER.GetGPREGRET() | 0xB1) // reset to DFU
		arm.SystemReset()
		return
	}
	if nrf.CLOCK.GetLFCLKRUN_STATUS() == 0 { // some bug they mentioned
		nrf.CLOCK.SetLFCLKRUN_STATUS(1)
	}

	nrf.WDT.CONFIG.Set((nrf.WDT_CONFIG_HALT_Pause << nrf.WDT_CONFIG_HALT_Pos) | (nrf.WDT_CONFIG_SLEEP_Run << nrf.WDT_CONFIG_SLEEP_Pos))
	nrf.WDT.CRV.Set(16 * 32768)                // 16 sec. timout
	nrf.WDT.RREN.SetBits(nrf.WDT_RREN_RR0_Msk) //Enable reload register 0
	nrf.WDT.TASKS_START.Set(1)
	for {
		time.Sleep(time.Second * 15)
		nrf.WDT.RR[0].Set(nrf.WDT_RR_RR_Reload) //Reload watchdog register 0
	}

```

Do you have any idea why or how? Thanks!

fanoush, 2022-06-06:
describe what `not really working` means

btw, you should clear reset reason you don't care for anymore, it is cumulative, nobody clears it for you automatically

vendelin8, 2022-06-08:
`not really working` means that I can't access the device any more, any I can dfu only from serial when the battery dies.

About clearing `GPREGRET`, I couldn't find any example stating that. And if I would, what should be set? 0 For all bits?

fanoush, 2022-06-08:
and btw the code you pasted - if it works perfectly like designed it would ping the watchdog and do nothing so there is really no way out of it as it is now. So you could ping watchdog only if button is not held or device is not on charger or something so that you could actually trigger watchdog and DFU.

vendelin8, 2022-06-08:
I already opened one. There's a red LED on my CH340G, that was on when it was connected to ds-d6 only. Now it's on only when I plug it into my laptop. I thought I broke the ds-d6 somehow, but now I'm sure that CH340G is wrong somehow. I think it's not just the LED, because since that, I couldn't program the ds-d6, that's why I opened it. Never mind, I already ordered some more CH340G-s locally, they are cheap.

Yeah, that makes sense to get a SWD debug set. If you don't need them, I'd buy a set from you, paypal works for me.

I have a nRF52832 bluetooth dongle I could use for testing. I can program it with SWD debugger only, right?

So please pack a small set just enough to SWD program these things, and let me know how much and your paypal. Revolut also works for me if you like.

And thanks.