fanoush/ds-d6 #18: Merged with main Espruino - Question

brendena, 2023-11-28:
This project is fairly amazing with the amount of devices it support.  I'm just curious why its not merged into the main Espruino repo.
https://github.com/espruino/Espruino

Is there any technical problems with merging it.

fanoush, 2023-11-29:
> Is there any technical problems with merging it.

what can be merged is already merged (see https://github.com/espruino/Espruino/pulls?q=is%3Apr+author%3Afanoush+is%3Aclosed+ ), what is here are board files and binary builds. Gordon does not want random board files in his repo as it is extra maintenance burden for him.  Also some patches are controversial or not clean enough for him so those are not in upstream repo - one example is when same SPI interface is shared between SPI flash and LCD - jeffmer maintains that one (lcd_spi_unbuf driver with shared SPI) , upstream has only minimal patches to make it work -  the `SPIFLASH_SHARED_SPI` bit https://github.com/espruino/Espruino/pull/1943/files 

overall for most devices if you want to build from source you take board file  from here or other places and build it with upstream source and it should  basically work. If not ask in Other boards in Espruino forum. if you don't have swd debugger and are afraid of bricking the watch it is better to ask first or use some other nrf52 devboard (or one piece of the watch you take apart) before uploading

brendena, 2023-12-02:
Ok that makes perfect sense.  Thank you.

closed 2023-12-02 by brendena