fanoush/ds-d6 #19: P8 display support

brendena, 2023-12-02:
Whats the status of P8 display support on the pinetime.
So experuino support [st7789](https://github.com/espruino/Espruino/blob/eabd13cfb8ce1c0a81708bb6e9f37f15b101a5fd/libs/graphics/lcd_st7789_8bit.h#L26
).

But i see you have LCD_SPI disabled and the pins commented out [are not configured ](https://github.com/fanoush/ds-d6/blob/master/espruino/DFU/P8/P8-SDK14.py#L112).

I know there's the "issue" with the spi hardware being used for both flash chip and the screen which might alter configuration.  The Espruino also setup for 8 bit display which i don't think is a issue, but i'm not entirely sure.

I did try you really [cool inlince C display program](https://gist.github.com/fanoush/3dede6a16cef85fbf55f9d925521e4a0) with this build [espruino_2v08.189_p8_SDK12_SD30.hex](https://github.com/fanoush/ds-d6/blob/master/espruino/DFU/P8/espruino_2v08.189_p8_SDK12_SD30.hex), but it didn't seem to work.  The interpreter throw a lot of errors.  

Also i just want to say this is a amazing project.  Thanks for making it.

fanoush, 2023-12-02:
> Espruino also setup for 8 bit display which i don't think is a issue, but i'm not entirely sure.

P8/Pinetime uses SPI interface for this display. 8 bit mode is for Bangle 1 - it does not use SPI but 8 data wires - different mode, that driver is not usable for us.

> I did try you really [cool inlince C display program](https://gist.github.com/fanoush/3dede6a16cef85fbf55f9d925521e4a0) with this build [...]

Enable pretokenization and/or minification otherwise it runs out of memory. Also reducing bitdepth - [line with bpp=4 to 1 helps](https://gist.github.com/fanoush/3dede6a16cef85fbf55f9d925521e4a0#file-p8-spilcd-demo-js-L516), but it should work with 4 too when uploaded code is tokenized/minified. The video in this post https://forum.espruino.com/conversations/347237/  is P8 running that code.

My display driver is fast because it uses in memory buffer - it uses lot of memory based on bit depth (the `Graphics.createArrayBuffer(240,240,bpp);` below the `bpp=` line) , alternative is jeffmer's spi_lcd_unbuf modification for shared SPI, he uses it in his own builds https://github.com/jeffmer/WatchApps It is slower, but keeps more memory available, which may make a lot of difference for P8. For nrf5840 watches (with 256KB ram instead of 64) is is not such issue. But it depends, enaon used my driver in his eucWatch https://github.com/enaon/eucWatch for P8 and it worked for him so iti s not that bad.

brendena, 2023-12-03:
Thanks for the clarification.  I am curios as to why you made this as a JavaScript inline c driver vs just a standard c file in the espruino project.  It seems like it would be very hard to make this driver?  Is this related to not wanting to clutter the main [espruino repo](https://github.com/fanoush/ds-d6/issues/18)?

So i was able to get it flashed on and i was able to turn on my screen.  But its weird my display just displays random noise.  I had to set the bpp to 2 and turn on the optimizations.  It might not work with bpp 4 for me because i'm using the newest espruino build 2.19.86

fanoush, 2023-12-03:
> because i'm using the newest espruino build 2.19.86

Oh, that explains it, please replace E.toString to E.toFlatString here
https://gist.github.com/fanoush/3dede6a16cef85fbf55f9d925521e4a0#file-p8-spilcd-demo-js-L462
there is broken compatibility in E.toString, it no longer returns flat string, see
https://github.com/espruino/Espruino/commit/2dc6905f66e8c2a1565c4c49798ad0d8a3a19345

> I am curios as to why you made this as a JavaScript inline c driver vs just a standard c file in the espruino project.

It was much easier and faster to develop as with Espruino native C code you need to build/flash/run/fix/build/flash/.....
With inlineC you iterate much faster - no firmware building and flashing to device repeatedly

brendena, 2023-12-04:
So that made it get closer, but it was causing so many interrupt 25 errors and i wanted to see this work.  So i flashed the version you provided and your display driver works!

Now my lofty goal would be to make my own "jswrap_bangle system".  To accomplish this I'll need to create a c driver that included in the espruino project.  So my thought would be to migrate some of this code into this driver.   Do you see any problems with this.

closed 2023-12-22 by brendena