fanoush/ds-d6 #20: How to update Espruino on Magic3/Rock?

projectitis, 2024-01-21:
Hi there,

I've [followed steps 1-6 here](https://github.com/fanoush/ds-d6/blob/master/espruino/DFU/Magic3/README.md).
I can now connect the Espruino IDE fine and upload apps etc.

However, I'm not sure how to update Espruino. Step 7 says:
`Install newer version of Espruino from this folder or use different package to install other custom software.`

Do I do this using DaFlasher again (following steps 1-4), or is there now a different way?

Thank you!

fanoush, 2024-01-22:
Yes, DaFlasher or nrfConnect for Android, last bullet of step 6 should switch device to Nordic DFU  mode and both tools should work.
In fact DaFlasher autodetects Espruino and can switch device to bootloader (send the poke command as mentioned on that page). In DaFlasher don't forget to check the nordic bootlader checkbox. I prefer nrfConnect as DaFlasher can be a bit temperamental but it depends so try the other if one of them does not work. nrfconnect after connecting to DfuTarg device should show Nordic Secure DFU service and DFU icon in top right corner.

projectitis, 2024-01-22:
Thank you! Can I also use vrfConnect or DaFlasher to install custom firmware created, for example, using c/c++ via platformIO + Arduino framework? The resulting files are `firmware.elf` and `firmware.hex`, not a `bin` file.

fanoush, 2024-01-23:
> Can I also use nrfConnect or DaFlasher to install custom firmware created, for example, using c/c++ via platformIO + Arduino framework? The resulting files are `firmware.elf` and `firmware.hex`, not a `bin`

No, unless it uses same SoftDevice version and you make DFU package out of your application hex file. Also if you make some mistake in your aplication and won't allow switching to bootloader you need to wait until battery dies.

You can have softdevice+bootloader version downgrade/upgrade DFU packages but this is quite advanced topic too. Check your arduino/platformio softdevice version, if it is S140 6.0 or 6.1 then it can work with stuff  I already have there.

> Also, when using nrfConnect (on ios) I get a list of devices - one of which is DfuTarg and one is Magic3 xxxx - but neither of them allow me to connect/pair. 

Only one works at the same time, probably some cache. You are not supposed to pair  connect in operating system like keyboard or headset, that can even cause more troubles. If you did that, delete it. Just scan for devices in nrfconnect and connect from that if you see it in the list. However I don't have any iOS device so not sure how it works.