fanoush/ds-d6 #21: flashing bin files (like dsd6-flash.bin) using black magic probe

aguaviva, 2024-03-27:
(Note more than a bug this is a note that hopefully will help others)

You can use the following command:
```
>arm-none-eabi-objcopy -I binary -O elf32-little --change-section-address .data=0x0000000 dsd6-flash.bin a.elf
```

fanoush, 2024-03-28:
cool, but this is not actually describing "flashing bin files using black magic probe" at all :-)

also depending on what is inside the elf file the `--change-section-address .data=0x0000000` may not be the right one or maybe it simply does nothing. For nrf52 both data and code sections go to different addresses than 0 so using that value is strange. For data it goes to RAM which is somewhere above 0x20000000 (depends on bluietooth stack configuration) and code goes to different places too, e.g. binaries built for SDK11 are linked to 0x1c000 address, SDK12 is 0x1f000, SDK14 is 0x23000, SDK15 is 0x26000.

also for flashing the hex file format instead of bin may be better since it has right address inside it so there is less change to write it to wrong place and also it can contain more parts in one file - like softdevice + application + bootloader + bootloader settings + uicr settings each one going to different address with 'holes' between

basically something like  `$(OBJCOPY) -O ihex $(PROJ_NAME).elf $(PROJ_NAME).hex` should work to make hex file

However since you actually did not write anything about blackmagic probe and which tool you use for flashing (openocd?) it is hard to guess what is the best way, openocd can handle hex file just fine - `program file.hex` is enough to flash.

fanoush, 2024-03-28:
Oh sorry, I get it now, I got it backwards at first, you are making elf file from bin file! not bin from elf
Oh, well, OK, but why? what it the flashing step you use with the elf file?

aguaviva, 2024-03-28:
I had a bricked tracker, and your original firmware in bin format. BMP wouldn't allow me to load bin files, but it allows me to flash elf, so that is why I converted it from bin to elf.

Glad to know you are still interested in this project :)