fanoush/ds-d6 #22: Question: Modifications for Magic 3

Chriz76, 2024-06-05:
Thank you for all your great work! 

I am trying to understand better how you adopted espruino to Magic 3 and I hope you can help me.

On a new Magic 3 the DaFit software ist running. I guess therefore DaFlasher is needed for flashing espruino_2v10.102_magic3-dafit.bin?

espruino_2v10.102_magic3-dafit.bin will replace the DaFit software on the watch. What is special about the espruino_2v10.102_magic3-dafit.bin? Does it use another SDK as the official espruino builds? Maybe because of a different SoftDevice on the Magic 3 or also because of special hardware? Do you have the sources/build.py for the espruino_2v10.102_magic3-dafit.bin?

Same for the Magic3-bootloader.txt. Is it the standard espruino bootloader? Or is there anything special about it and do you have the sources?

Thank you so much!

fanoush, 2024-06-06:
The bin version was a bit special because at that time espruino had some bugs when runnning with softdevice S140 6.1.x, also some stuff was disabled to fit the DaFit size limit as described in Magic3 folder README. Nowadays I think the upstream version would work as is except maybe the size.

bootloader is standard espruino bootloader again with some slight fixes - it does time out and won't stay in dfu forever, it can be switched to dfu mode by poking GPREGRET register from espruino.

I can provide patch for the bootloader and updated board file to rebuild from recent upstream source.

Chriz76, 2024-06-06:
Thank you! The patches would be great. 

Here is a little background.

I am still on my quest to create a full waterproof espruino. That means buttonless mode and I already checked the different code parts that I have to change like bootloader main.c, peripheralpollhandler, etc. so soft and hard reset, shutdown, dfu, etc are still working. And that will be some places.

For the start I will use the Magic 3 to test opening the watch, removing the button, resealing the watch and testing the water resistance. And also the software changes. So if anything breaks it is only 10$ and not 120€. Though I also try to get a used bangle js 2 for the final steps.

Hopefully in the end it might be working for me and the community, too. With the code, a tutorial or even a 3D print buttonless case. Many people want to wear their watches in the shower or while swimming. And if you want to do it cheap then this means buttonless.

Chriz76, 2024-06-07:
Your image is now running on the Magic3!

[image]

Dissassembly was easy but the button can't be removed and so I will have to glue it from outside.

But first I have to change the bootloader to buttonless. So looking forward to the magic3 espruino bootloader code.

fanoush, 2024-06-10:
Here is minimal patch for MAGIC3
[magic3.patch](https://github.com/user-attachments/files/15774989/magic3.patch)
clone Espruino git tree and apply via 'git apply file' and then build
This should work for ubuntu 22.04
```
sudo apt-get install build-essential git curl python-is-python3 python3-pip
pip3 install nrfutil
#now maybe refresh path by logout/login so nrfutil is found in ~/.local/bin
git clone https://github.com/espruino/Espruino
cd Espruino/
git apply ../magic3.patch
. scripts/provision.sh MAGIC3 #this tries to install nrfutil via 'sudo pip install nrfutil' if not found, which is not that great
make BOARD=MAGIC3 RELEASE=1 clean
make BOARD=MAGIC3 RELEASE=1
```
I have more changes in main Espruino but this one is minimal and should be enough
This can't work as  -dafit.bin, that one is limited to 0x3f000 size while according to `hexinfo.py bin/espruino_2v22.17_magic3.hex` the size is 0x000435E8

BTW if you have the device opened you can use openocd and flash the full hex via SWD, otherwise build via `make BOARD=MAGIC3 RELEASE=1 DFU_UPDATE_BUILD=1` to make espruino_2v22.17_magic3.zip for DFU

Chriz76, 2024-06-10:
Thank you so much. I will try it out!

closed 2024-06-10 by Chriz76