fanoush/ds-d6 #23: Waiting for INIT DFU notification timesout

aguaviva, 2025-04-13:
Thanks for this awesome tool chain :) I know its been a while but was wondering if you'd know why I am getting this. Flashing seems to work (after many attempts) with DaFlasher. I'm using the bootloader with the fixed arrows.

Here is the log:

```
(py27) : ~/repos/nrf52833/$ python2.7 ./desay_dfu.py -z /tmp/arduino_build_434041/D6Emulator.ino.zip -a D3:71:90:1C:E9:C8 -v
Sending file D6Emulator2.ino.bin to D3:71:90:1C:E9:C8
Binary imge size: 45588
Binary CRC32: 1063361471
scan_and_connect
Connecting to D3:71:90:1C:E9:C8
Connected
Checking DFU State...
Trying to switch to DFU mode
Found Desay firmware, rebooting to bootloader
scan_and_connect
Connecting to D3:71:90:1C:E9:C9
Connected
Connected True
Checking DFU State...
Control Point Handle: 0x0010, CCCD: 0x0011
Packet handle: 0x000e
Enabling notifications
_enable_notifications
char-write-req 0x0011 0100
Sending START_DFU  4 [0, 0, 0, 0, 0, 0, 0, 0, 20, 178, 0, 0]
_dfu_send_command
char-write-req 0x0010 0104
char-write-cmd 0x000e 000000000000000014b20000
Waiting for Image Size notification
Waiting for notification
dfu_wait_for_notify
Parsing notification
['10', '01', '01']
opcode: 0x10, proc: START_DFU, res: SUCCESS
_dfu_send_command
char-write-req 0x0010 0200
dfu_send_init
char-write-cmd 0x000e ffffffffffff00000100feff9820
_dfu_send_command
char-write-req 0x0010 0201
Waiting for INIT DFU notification
Waiting for notification
dfu_wait_for_notify
Exception at line 954: No notification received
DFU done
```

Many thanks!

aguaviva, 2025-04-14:
serial updating works like a charm though, many thanks for this :)

fanoush, 2025-04-15:
Hi, I think I did implement some retry/restart for such errors but I guess it was just my local change I did not push and I am not sure I will still find it, but I'll check. However I am not sure, maybe the last commit https://github.com/fanoush/ds-d6/commit/688628b0af01f65d1c69a77b141aede2b25d8dcb was already it as it looks similar to what I remember.

What happens if you run it again but with the +1 address of the bootloader (D3:71:90:1C:E9:C9) in this state? It is switched to bootloader so the original address won't work but the bootloader one should and hopefully it will eventually succeed.

aguaviva, 2025-04-15:
Hi Fanoush!  tried that, put the tracker into bootloader mode but got the following:

```
(py27) $  desay_dfu.py -z /tmp/arduino_build_55986/D6Emulator2.ino.zip -a D3:71:90:1C:E9:C9 
Sending file D6Emulator2.ino.bin to D3:71:90:1C:E9:C9
Binary imge size: 45588
Binary CRC32: 2085683641
Connecting to D3:71:90:1C:E9:C9
Couldn't connect, will try DFU MAC
Connecting to D3:71:90:1C:E9:CA
Exception at line 964: Can't connect to device
DFU done
```

fanoush, 2025-04-16:
strange, it should connect, but now I see it tries the +1 mac too so it is not needed to change the mac and should connect even when run with the normal D3:71:90:1C:E9:C8 one. I still have some DS-D6 somewhere, will test.

aguaviva, 2025-04-16:
Hold on, you were very right (I have 2 trackers and used the wrong one!). Using mac+1 in bootloader-mode did flash the tracker, at the end though it gets stuck in bootloader mode and the script timesout.

```
(py27) $ desay_dfu.py -z /tmp/arduino_build_55986/D6Emulator2.ino.zip -a c5:9d:57:d2:a6:84
Sending file D6Emulator2.ino.bin to C5:9D:57:D2:A6:84
Binary imge size: 39224
Binary CRC32: 2684661890
Connecting to C5:9D:57:D2:A6:84
Connected
Waiting for Image Size notification
Waiting for INIT DFU notification
Begin DFU (MTU 23, packet size 20)
Progress: |xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx| 100.0% Complete (39224 of 39224 bytes)

Upload complete in 0 minutes and 26 seconds
Waiting for DFU complete notification
Waiting for Firmware Validation notification
Exception at line 956: No notification received     <-------------------times out
DFU done
```

fortunately I can flash using serial 
```
$ adafruit-nrfutil dfu serial -p /dev/ttyUSB0 -pkg /tmp/arduino_build_55986/D6Emulator2.ino.zip -b 38400
```