fanoush/ds-d6 #24: Issues uploading Espruino DFU on the P8

smolbun, 2025-04-21:
1. I've installed [DaFitBootloader23Hacked.bin](https://github.com/atc1441/DaFlasherFiles/blob/master/DaFitBootloader23Hacked.bin) through DaFlasher, the color seems to be inverted and screen is mirrored.
2. I then installed espruino_2v05.31_p8_SDK14_SD50.zip, but there is no screen at all now. I'm able to connect to it through the Espruino IDE.
3. I tried uploading others through "Do DFU Update", but I kept getting GATT ERROR in DaFlasher (Use Nordic Bootloader checked):
espruino_2v08.5_p8_SDK12_SD30_SPIFLASH.zip
espruino_2v10.215_p8_SDK12_SD30_SPIFLASH4M.zip

Clicking on "Start Bootloader" seems to not do anything.
[image]

fanoush, 2025-04-21:
Hello. Blank screen is normal, you would need to load some javascript code to use the screen. DaFlasher should work in this mode, try again, toggle bluetooth on the phone.

You cannot flash SDK12 binaries when you are on SDK14, you need to flash downgrade to SDK11 package and then flash upgrade SDK11 to SDK12 package and then SDK12 binaries would work.

smolbun, 2025-04-21:
Hi thank you for taking the time to reply, I ended up soldering  the swd pins to my st-link and got ATCwatch flashed instead. 

Not exactly sure what's the difference between this two but I've selected this:
[image]
After clicking "Burn bootloader" the tux penguin appeared which I then click upload to get the firmware running.

Now I'm not sure how to flash [DaFitBootloader23Hacked.bin](https://github.com/atc1441/DaFlasherFiles/blob/master/DaFitBootloader23Hacked.bin) since DaFlasher takes me into DFU mode. How can I flash the new bootloader over bluetooth and through swd?

Apologies as this is my first time messing with these.

fanoush, 2025-04-22:
> Now I'm not sure how to flash [DaFitBootloader23Hacked.bin](https://github.com/atc1441/DaFlasherFiles/blob/master/DaFitBootloader23Hacked.bin) since DaFlasher takes me into DFU mode.

You can't, but why? what are you trying to achieve by that? That binary is custom format for DaFit app.

> How can I flash the new bootloader over bluetooth and through swd?

over bluetooth via compatible zip file (see previous post), for swd the easiest is hex file

bootloader is paired with compatible softdevice so you typically update both via one zip file and both together are specific to nordic SDK you use for building the main application. Arduino for P8 is SDK11, espruino can be SDK11,12,14,15 - for P8/nrf52832 the best for Espruino is SDK12

> After clicking "Burn bootloader" the tux penguin appeared which I then click upload to get the firmware running.

sorry, no idea what that means, for Arduino/ATCwatch better create issue there or join atc1441 discord https://discord.gg/CmVE8UzX

smolbun, 2025-04-22:
Oh I didn't know that the bootloader is different for every firmware, if I understood correctly that means the bootloader is in the same .bin / .hex as the Espruino firmware right?

I've uploaded `p8-sdk11-to-sdk12.zip` .

Not sure what is the CV16 for, I believe I have the P8b variant, does it affect the flashing? I'm having issues uploading anything other than boot.

1. `espruino_2v10.215_p8_SDK12_SD30_SPIFLASH4M.zip`
2. `espruino_2v10.158_p8_SDK12_SD30.zip`
3. `espruino_2v10.158_p8_SDK12_SD30.zip` somehow I got main to install but not clock (`p is not defined`), still no display though.

[image]

sometimes I get this error :
[image]

fanoush, 2025-04-22:
> Oh I didn't know that the bootloader is different for every firmware, if I understood correctly that means the bootloader is in the same .bin / .hex as the Espruino firmware right?

No. Bootloader is the thing that does the DFU mode = allows to upload zip package with application firmware or another bootloader+softdevice over bluetooth. With no bootloader only SWD works.

> 3\. espruino_2v08.189_p8_SDK12_SD30.zip somehow I got main to install but not clock (`p is not defined`), still no display though.
> 
> [image]

If you are trying jeffmer's stuff  you need to flash his espruino build as per https://github.com/jeffmer/WatchApps?tab=readme-ov-file#p8

smolbun, 2025-04-22:
I've tried jeffmer's [espruino_2v11.222_p8.zip](https://github.com/jeffmer/WatchApps/blob/master/firmware/espruino_2v11.222_p8.zip). Had a brief moment of success, I was able to install boot and main but when I tried launcher it kept disconnecting. The display seem to work though it was inverted and flip (probably because its a p8b), I tried reflashing with jeffmer's p8 firmware and this time I couldn't get boot installed, it just stops half way and disconnects. Guess I'm out of luck with this variant.

fanoush, 2025-04-22:
> Guess I'm out of luck with this variant.

inverted/rotated display is just fix of two or three numbers in display initialization array
you can search jeffmer's code for somethings similar to this https://gist.github.com/fanoush/3dede6a16cef85fbf55f9d925521e4a0#file-p8-spilcd-demo-js-L496 to change MADCTL rotation settings and add/remove command 0x21 down below, for more info about commands google ST7789 datasheet

as for disconnects it could be that you are either doing something wrong (like uploading to RAM instead of storage) or it can be related to some IDE settings (minification, tokenization) or your computer simply has bad/noisy BLE connection for some reason.

smolbun, 2025-04-23:
[espruino_2v08.189_p8_SDK12_SD30.hex](https://github.com/fanoush/ds-d6/blob/master/espruino/DFU/P8/espruino_2v08.189_p8_SDK12_SD30.hex) I was able to upload "Boot" and "Main" through the app loader but doing so for "Launcher" would result in `p is not defined`, I could view the files uploaded in the IDE. When I upgraded to espruino_2v11.222_p8.zip the display sort of work.

Since I still could not install anything via the app loader, I deleted all the files from the earlier firmware and flashed it again.
[espruino_2v11.222_p8.zip](https://github.com/jeffmer/WatchApps/blob/master/firmware/espruino_2v11.222_p8.zip)
I tried uploading a simple `console.log("test");` to flash but I can't seem to save it into flash. Neither can I upload anything through the app loader without being disconnected.

[image]

fanoush, 2025-04-24:
Oh, maybe I know. For the first time you need to wake the SPI flash chip as per https://github.com/enaon/eucWatch?tab=readme-ov-file#manual-scripts-install since the original dafit app put it to sleep. That is why any version with storage in external flash ends with jshFlashWrite errors.

smolbun, 2025-04-24:
1. I ended up doing the long winded method of [installing eucwatch](https://enaon.github.io/eucWatch/tools/hackme/) then clicking "Install all apps" which inits the flash chip.
2. Opened IDE, clear everything out.

Was I supposed to just execute this `var spi=new SPI();spi.setup({sck:D2,mosi:D3,miso:D4,mode:0}); spi.send([0xab],D5); //wake` in the IDE under RAM?

3. Anyways, I installed [espruino_2v11.222_p8.zip](https://github.com/jeffmer/WatchApps/blob/master/firmware/espruino_2v11.222_p8.zip) again.

https://jeffmer.github.io/WatchApps/ seems to be extremely buggy, I was able to install "Boot", "Main", "Launcher (Default)" after several attempts but not "Settings". The error I usually get was: `<BLE> Disconnected (gattserverdisconnected)`. 

What would be the proper way of installing the apps manually?  Do I just copy everything over? For example "Settings" https://github.com/jeffmer/WatchApps/tree/master/apps/setting. I notice there was files with .info when I was in the IDE, what are they for?

4. Edit `lcd-p8.js` set `INVERSE=0`

[image]

I did send `reset run` from openocd but it didn't seem to fix it.

fanoush, 2025-04-24:
> Was I supposed to just execute this `var spi=new SPI();spi.setup({sck:D2,mosi:D3,miso:D4,mode:0}); spi.send([0xab],D5); //wake` in the IDE under RAM?

yes

> 7\. Edit `lcd-p8.js` set `INVERSE=0`
> 
> [image]

so inversion looks fixed, now rotation - 0x36 command, and if it will be still off after rotation (I guess it won't) then the offset - 0x37 command

as for the disconnects and manual installation I don't know, I'd try the apploader again and possibly put the watch near the computer or open the apploader from web browser on android phone, I think it does some magic that is not that easy to do manually

smolbun, 2025-04-24:
I changed it to `cmd(0x36, 0x48);` instead now it looks fine.

After lots of attempts of disconnecting the Bluetooth and uploading the "Settings" app, it finally uploaded. I don't think it's a device issue since I've tested with 2 PCs and my Android phone.

Last few questions, is it possible to use the "Android Integration" from https://banglejs.com/apps/ ? I need some way to get notifications and call Bangle.http.

[image]

fanoush, 2025-04-24:
> I don't think it's a device issue since I've tested with 2 PCs and my Android phone.

what enaon does is to replace the antenna wire in basically every P8 he used, he said he opened many and the lenght of the wire is pretty random and mostly wrong. see also point 2. at https://github.com/enaon/eucWatch/

> Last few questions, is it possible to use the "Android Integration" from https://banglejs.com/apps/

possibly yes, ask also jeffmer,  I think he has some version of it there. The issue is that it is old an unmaintained and there is no 100% compatibility with real Bangle, but everything is possible with enough effort